Olen/solar-monitor #2: Great monitor!

randrcomputers, 2021-12-19:
I have this working with Renogy controller and works great! thank you!
I see it can also monitor batteries BMS but i have Xiaoxiang BMS will this work with that BMS? I did try to connect to the batteries but failed. could a plugin be added for Xiaoxiang?

Olen, 2021-12-20:
Hi,
I am glad to hear it works for you.
Regarding plugins, it can probaly be done if it uses BLE.  However, "somebody" need to decode the protocol it uses, and add the appropriate functions.

oedo808, 2021-12-20:
@Olen, you might consider updating the documentation that this works with controllers compatible with the Renogy BT-1 module.  It appears to be the same as SolarLink for the most part.  We were only able to find this project by searching for some of the BT-1 UUIDs published and noticed that it works great!  I can submit a PR to add some of the Renogy documentation based on my usage.

Olen, 2021-12-21:
Updated the README now.

Also committed a few minor fixes that was running in my "production" environment, but had not been pushed here.

Anto79-ops, 2021-12-28:
yes, great monitor! Also got it working with my BT-1 device. Anyway you can increase the decimal places/significant digits of the current values? for example, it would be nice to see the value as 0.43 vs 0.4 amp.

hamanjam, 2022-02-06:
Has anyone been successful reading a Rover Elite 40 with BT-2? Reading through the thread, I can't tell what success people have been having with BT-2. I can connect but it immediately disconnects and I get no results. I see a lot of "ping" requests every couple minutes but no data. I'm a Linux guy but not at all python. Here's what I get:
pi@inventorpi:~/solar-monitor $ ./solar-monitor.py
Debug enabled
DEBUG: Creating new DataLogger
DEBUG: Creating new MQTT-logger
DEBUG: MQTT Sending CONNECT (u0, p0, wr0, wq0, wf0, c1, k60) client_id=b'localhost'
DEBUG: MQTT Received CONNACK (0, 0)
INFO: Adapter status - Powered: True
INFO: Starting discovery...
DEBUG: Found 15 BLE-devices so far
DEBUG: Found 20 BLE-devices so far
DEBUG: Found 20 BLE-devices so far
DEBUG: Found 20 BLE-devices so far
DEBUG: Found 20 BLE-devices so far
DEBUG: Found 20 BLE-devices so far
INFO: Found 20 BLE-devices
DEBUG: Processing device <hidden MAC> None
DEBUG: Processing device <hidden MAC> <hidden>
DEBUG: Processing device <hidden MAC> <hidden>
DEBUG: Processing device <hidden MAC> <hidden>
DEBUG: Processing device <hidden MAC> <hidden>
DEBUG: Processing device <hidden MAC> <hidden>
DEBUG: Processing device <hidden MAC> None
DEBUG: Processing device <hidden MAC> <hidden>
DEBUG: Processing device <hidden MAC> <hidden>
DEBUG: Processing device f0:f8:f2:6e:86:0b BT-TH-F26E860B
INFO: Trying to connect to f0:f8:f2:6e:86:0b...
INFO: Successfully imported SolarLink.
DEBUG: New PowerDevice
DEBUG: New RegulatorDevice
INFO: [regulator] Connecting to f0:f8:f2:6e:86:0b
DEBUG: Processing device <hidden MAC> None
DEBUG: Processing device <hidden MAC> None
DEBUG: Processing device <hidden MAC> <hidden>
DEBUG: Processing device <hidden MAC> <hidden>
DEBUG: Processing device <hidden MAC> <hidden>
DEBUG: Processing device <hidden MAC> None
DEBUG: Processing device <hidden MAC> <hidden>
DEBUG: Processing device <hidden MAC> None
DEBUG: Processing device <hidden MAC> <hidden>
DEBUG: Processing device <hidden MAC> <hidden>
INFO: Terminate with Ctrl+C
INFO: [f0:f8:f2:6e:86:0b] Discovered, alias = BT-TH-F26E860B
INFO: [regulator] Connected to BT-TH-F26E860B
INFO: [f0:f8:f2:6e:86:0b] Discovered, alias = BT-TH-F26E860B
INFO: [regulator] Disconnected
DEBUG: MQTT Sending PINGREQ
DEBUG: MQTT Received PINGRESP

---------------
Here's my ini:

[monitor]
adapter = hci0
debug = True
temperature = F
# C = Celsius
# K = Kelvin
# F = Farenheit

[datalogger]
#url = http://server/solar/api/
#token = 39129e20be0503937cb72a5f719337cc

[regulator]
type = SolarLink
mac = f0:f8:f2:6e:86:0b
reconnect = True

[mqtt]
broker = localhost
prefix = solar
username = user
password = password
hostname = localhost

hamanjam, 2022-02-08:
I did find if I install "bluez-hcidump", I get a lot of data in debug mode:

DEBUG: [regulator] Writing data to 0000ffd1-0000-1000-8000-00805f9b34fb - bytearray(b'main recv da ta[ff] [') (6d61696e20726563762064612074615b66665d205b)
DEBUG: REG: BatteryParamInfo VAL: b'\xff\x03\x0e\x00P\x00\x7f\x00\x00\x15\x10\x00\x00\x00\x00\x00\x00E2'
DEBUG: mSOC 0 80 => 80 %
DEBUG: [regulator] Value of _dsoc changed from 0 to 800.0
DEBUG: mVoltage 0 127 => 12.700000000000001 V
DEBUG: [regulator] Value of _charge_mvoltage changed from 0 to 12700.000000000002
DEBUG: mElectricity 0 0 => 0.0 A
DEBUG: [regulator] Value of _charge_mcurrent out of bands: Changed from 0 to 0.0 (no diff)
DEBUG: mDeviceTemperature 21
DEBUG: [regulator] Value of _dkelvin changed from 2731 to 2941.0
DEBUG: mDeviceTemperatureCelsius 21.0
DEBUG: mBatteryTemperature 16
DEBUG: [regulator] Value of _bkelvin changed from 2731 to 2891.0
DEBUG: mBatteryTemperatureCelsius 16.0
DEBUG: mLoadVoltage 0 0 => 0.0 V
DEBUG: [regulator] Value of _mvoltage out of bands: Changed from 0 to 0.0 (no diff)
DEBUG: mLoadElectricity 0 0 => 0.0 A
DEBUG: [regulator] Value of _mcurrent out of bands: Changed from 0 to 0.0 (no diff)
DEBUG: mLoadPower 0 0 => 0 W
DEBUG: [regulator] Value of _mpower out of bands: Changed from 0 to 0.0 (no diff)
INFO: [regulator] Sending new data current: 0.0
DEBUG: MQTT Sending PUBLISH (d0, q0, r0, m1), 'b'homeassistant/sensor/regulator/current/config'' (NULL payload)
DEBUG: Published to MQTT
DEBUG: [regulator] Looping thread Device-poller-thread regulator
DEBUG: [regulator] Looping thread Device-poller-thread regulator
DEBUG: create_poll_request SolarPanelInfo
DEBUG: create_poll_request SolarPanelInfo => [255, 3, 1, 7, 0, 4, 225, 234]
DEBUG: [regulator] Writing data to 0000ffd1-0000-1000-8000-00805f9b34fb - [255, 3, 1, 7, 0, 4, 225, 234] (ff0301070004e1ea)
DEBUG: Creating MQTT-sensor solar/regulator/current/state
DEBUG: MQTT Sending PUBLISH (d0, q0, r1, m2), 'b'homeassistant/sensor/regulator/current/config'', ... (199 bytes)
DEBUG: Published to MQTT
DEBUG: Publishing to MQTT localhost: solar/regulator/current/state = 0.0
DEBUG: MQTT Sending PUBLISH (d0, q0, r1, m3), 'b'solar/regulator/current/state'', ... (3 bytes)
INFO: [regulator] Sending new data input_current: 0.0
DEBUG: Published to MQTT
DEBUG: MQTT Sending PUBLISH (d0, q0, r0, m4), 'b'homeassistant/sensor/regulator/input_current/config'' (NULL payload)
DEBUG: Published to MQTT
DEBUG: [regulator] Looping thread Device-poller-thread regulator
DEBUG: [regulator] Looping thread Device-poller-thread regulator

WWE-Corey, 2022-05-30:
> I did find if I install "bluez-hcidump", I get a lot of data in debug mode:
>
> DEBUG: [regulator] Writing data to 0000ffd1-0000-1000-8000-00805f9b34fb - bytearray(b'main recv da ta[ff] [') [...]

Did you ever get the BT-2 working (like the BT-1 does)?

Olen, 2023-04-27:
Just some spring cleanup of old issues.

closed 2023-04-27 by Olen