Olen/solar-monitor #7: How to support Renogy Battery w/ Bluetooth

itemir, 2022-02-09:
First of all, thank you for this repo. I have been looking for something like this for a long time and I am very happy to see it!

I am trying to use this with [Renogy Bluetooth Batteries](https://www.renogy.com/12v-100ah-lithium-iron-phosphate-battery-w-bluetooth/). Doesn't look like it is supported, which makes sense. I am wondering if you have any pointers on how to add a new plugin? I imagine this includes sniffing the traffic a bit but I wasn't sure. Any pointers in enabling this will be highly appreciated. Thanks!

acmagn, 2022-02-23:
Hey itemir,
I don't know where you are with this, but I am working on this on the side right now. It ended up being more different from a Renogy BT Controller than I originally thought. Yeah it ended up involving some bluetooth sniffing and deciphering the outputs. I already have some almost-stable code which I will try to get ready-to-share in about a week (haven't pushed anything yet). I am currently able to grab voltage, current, cell voltage, temperature, and working on remaining capacity.

I will comment here when I have pushed up something.

itemir, 2022-02-23:
That is awesome @acmagn ! Look forward to testing the changes.

acmagn, 2022-03-04:
Created a PR to provide the battery support.
I only have one type of the Renogy batteries (200AH) to test on, so curious to hear if it works smoothly for yours as well.

Anto79-ops, 2022-03-04:
do you think it will work if I place the mac address of BT-2 module (for battereis that use BT-2 module i.e. not built-in bluetooth?)

acmagn, 2022-04-26:
@Olen I think this issue can be closed as it seems the RenogyBatt plugin has reached a good level of stability.

WWE-Corey, 2022-05-21:
>  I could give you some details/assistance on how to sniff the Renogy DC Home app and figure out what changes are needed.

I have the same setup from issue #12 - how can I read/sniff the messages from the BT-2 to try and get this integrated into Solar Monitor?

Anto79-ops, 2022-05-21:
> > I could give you some details/assistance on how to sniff the Renogy DC Home app and figure out what changes are needed.
> 
> I have the same setup from issue #12 - how can I read/sniff the messages from the BT-2 to try and get this integrated into Solar Monitor?

I have a BT-2 module battery and this script works for this battery also.

WWE-Corey, 2022-05-21:
> I have a BT-2 module battery and this script works for this battery also.

I see readings in the log but I have not been able to get the Renogy battery to transmit over MQTT - as far as how to sniff, I get errors for the inverter/charger so need to try and get it figured out.

acmagn, 2022-05-25:
@WWE-Corey I'm going to give the longer answer, I hope this is clear enough. I'm not an expert, so bare with me. 
Easiest way for me was to enable Bluetooth capture on the device with the app -- an android phone for me, but pretty much any device will have guides online. Once you have bluetooth capture on, run the DC Home App and let it load your inverter's info. Then you have to look at the capture in Wireshark. (([here's a very simple guide to do this.](https://support.honeywellaidc.com/s/article/How-to-capture-Bluetooth-traffic-from-and-to-an-Android-Device))  
For interpreting the output, this is what I did:
1) use wireshark's filtering to see only the bluetooth communication between your inverter's BT address and your device.  If you're mostly interested in monitoring, focus on read requests sent from your device to the inverter.  (I believe the command is the second or third byte in the data sent: 03 for read, 06 for write, but Wireshark will give you this info as well).
2) Write down/log all unique read requests sent from your device to the inverter.
3) Use something like bluetoothctl menu gatt to test the read requests and see what they do. For some reason for me, I had to have bluetoothctl open in two terminals. One subscribed to the notify_service_UUID, the other to write to the write_service_UUID (see the plugins.md file that Olen made).
5) with the most promising read requests, I wrote a python script to convert the responses from base-16 to decimal and determined what each did from there. The part that got me hung up for a while was that the reading values varied in numeric length in the response (for example some were simply FF, others where FF 00 00). Converting FF and FF0000 to decimal looks incredibly different.
I hope this helps!

WWE-Corey, 2022-05-26:
> @WWE-Corey I'm going to give the longer answer, I hope this is clear enough. I'm not an expert, so bare with me. Easiest way for me was to enable Bluetooth capture on the device with the app -- an android phone for me, but pretty much any device will have guides online. Once you have bluetooth capture on, run the DC Home App and let it load your inverter's info. Then you have to look at the capture in Wireshark. (([here's a very simple guide to do this.](https://support.honeywellaidc.com/s/article/How-to-capture-Bluetooth-traffic-from-and-to-an-Android-Device)) For interpreting the output, this is what I did:
> 
> 1. use wireshark's filtering to see only the bluetooth communication between your inverter's BT address and your device.  If you're mostly interested in monitoring, focus on read requests sent from your device to the inverter.  (I believe the command is the second or third byte in the data sent: 03 for read, 06 for write, but Wireshark will give you this info as well).
> 2. Write down/log all unique read requests sent from your device to the inverter.
> 3. Use something like bluetoothctl menu gatt to test the read requests and see what they do. For some reason for me, I had to have bluetoothctl open in two terminals. One subscribed to the notify_service_UUID, the other to write to the write_service_UUID (see the plugins.md file that Olen made).
> 4. with the most promising read requests, I wrote a python script to convert the responses from base-16 to decimal and determined what each did from there. The part that got me hung up for a while was that the reading values varied in numeric length in the response (for example some were simply FF, others where FF 00 00). Converting FF and FF0000 to decimal looks incredibly different.
>    I hope this helps!

Thank you! I'll try to get started with this. I appreciate the time and effort you spent writing this up.

WWE-Corey, 2022-05-29:
I think I got part of it figured out. I was able to get a btsnoop_hci.log into wireshark and filter down from my phone to the BT-2. I then marked & commented all the "read" packets which I exported to a pcapng file. Unfortunately now at a loss as to what to do next.

[image]

acmagn, 2022-06-03:
@WWE-Corey Looks good so far. So the Read of "GATT characteristic declaration" is the app requesting metadata about the available attributes UUIDs. The device first gathers metadata and then later does read value requests based on them. You should also have Read Value requests in the btsnoop_hci.log. Chronologically, they'll be near the end. In wireshark you'll need to open up the details of these requests and get the actual raw data being written to the device. That raw data will be in the form of something like "48 03 FF 01 00 00 12 54 32"
I couldn't find a good summary, but this article talks about this: https://www.oreilly.com/library/view/getting-started-with/9781491900550/ch04.html
If you can't get it from there, I'll track down my btsnoop_hci.log (my old computer just died) and get some screenshots to show you what it looks like.

WWE-Corey, 2022-06-04:
> @WWE-Corey Looks good so far. So the Read of "GATT characteristic declaration" is the app requesting metadata about the available attributes UUIDs. 

I appreciate the help but it seems this is beyond my current capacity. I went back through the logs and believe I found these values but still do not comprehend what to do.

[image]

Olen, 2023-04-27:
Just some spring cleanup of old issues.

closed 2023-04-27 by Olen