Olen/solar-monitor #12: Interested in Helping using my BT-2, Hub, 48V 3500W SCC, 48v Batteries

excalq, 2022-04-06:
Hello, I'm the owner of an off-grid cabin with a newish setup of these items. For better or worse, I spent quite a bit going "all in" on Renogy, only discover nothing provides any sort of remotely usable telemetry. I'd love to provide assistance getting these devices "figured out" to hopefully create a plugin here. I am fluent in Python, Linux sysadmin, but a novice with electronics. I have a Raspberry Pi 4 on site, which I can access remotely when not on site.

## Devices Present
* [Renogy 48v 3500w Solar Inverter/Charge Controller](https://www.renogy.com/48v-3500w-solar-inverter-charger/)
   This is the device that I'd really like data from. It has a single RJ45 jack, labeled with RS485/CAN.
   Also appears to be whitelabelled by other brands: 
  * [EcoWorthy](https://www.eco-worthy.com/collections/solar-inverters/products/all-in-one-inverter-built-in-3500w-48v-pure-sine-wave-power-inverter-80a-mppt-controller-for-off-grid-system), 
  * [PowMr](https://powmr.com/inverter-charger/solar-inverter-charger/solar-inverter-charger/powmr-solar-inverter-charger-3500w-built-in-mppt-80a-solar-controller-max-pv-input-145v-dc-output-110-120vac-48vdc-system-3.5kw-pure-sine-wave-hybrid-inverter-suitable-us-ca-mx/)
  * [SRNE HF4830S60](https://www.srnesolar.com/product/solar-charge-inverter-hf4830s60) This may be the original OEM device?
* [Renogy 48v 50Ah battery](https://www.renogy.com/48v-50ah-smart-lithium-iron-phosphate-battery/)
    This also has a pair of RS485 and CANBUS RJ45 jacks, but the BT-2 does not read them, in my experience.
* [Renogy BT-2 Bluetooth Module](https://www.renogy.com/bt-2-bluetooth-module/)
* [Renogy Communication Hub](https://www.renogy.com/communication-hub/)

I've tried connecting Solar-Monitor to the BT-2 module, using various "plugins", and it does connect using _SolarLink_, but is not giving useful data out. I do get lines such as:

```
# [MACs obscured]
INFO: [e0:7d:ea:**:**:**] Discovered, alias = BT-TH-EA******
INFO: [inverter] Connected to BT-TH-EA******
INFO: [inverter] Resolved services
INFO: [inverter]  Service [f000ffd0-0451-4000-b000-000000000000]
INFO: [inverter]    Characteristic [f000ffd1-0451-4000-b000-000000000000]

DEBUG: create_poll_request BatteryParamInfo
DEBUG: create_poll_request BatteryParamInfo => [255, 3, 1, 0, 0, 7, 16, 42]
DEBUG: [inverter] Writing data to 0000ffd1-0000-1000-8000-00805f9b34fb - [255, 3, 1, 0, 0, 7, 16, 42] (ff0301000007102a)
DEBUG: [inverter] Write to characteristic done for: [0000ffd1-0000-1000-8000-00805f9b34fb]
DEBUG: [inverter] Looping thread Device-poller-thread inverter
DEBUG: [inverter] Looping thread Device-poller-thread inverter
DEBUG: create_poll_request SolarPanelInfo
DEBUG: create_poll_request SolarPanelInfo => [255, 3, 1, 7, 0, 4, 225, 234]
DEBUG: [inverter] Writing data to 0000ffd1-0000-1000-8000-00805f9b34fb - [255, 3, 1, 7, 0, 4, 225, 234] (ff0301070004e1ea)
```

But no data is written to MQTT. I'm happy to post `bluetooth show <MAC>` output as well.

Anyway, I'm happy to write some code and get these devices supported. Where would I begin? Even better if I can do some remotely via the Pi, as I'm usually four hours of driving away from these devices.

Olen, 2022-04-06:
One of the things I did when I started this project was to run an android emulator on my laptop and do a packet dump (wireshark) of the bluetooth-traffic when the original app connected.

You can also fetch the traffic directly from an android phone.  I believe this should still work, but maybe with some adjustments:
https://medium.com/propeller-health-tech-blog/bluetooth-le-packet-capture-on-android-a2109439b2a1

As you can see in your log, in that session we are first sending some data to the device.  This is done to initiate the transmission of data from the inverter.  I have no idea what the data is supposed to mean, it is just a mirror of what the original app was doing when it initiated a connection.
With your bluetooth logs/packet dumps you will probably see some similar patterns.  That it sends some data first, and then starts receiving.  For some of the devices, you might also need to send some kind of "keep alive" at regular intervals to avoid disconnects.

This should at least get you started.  

The plugins themselves are pretty "simple".  They read the data from the devices, parses them an converts it to common formats, and have a few common function names defined (I should probably document that at some time) where they provide the data in the right format.

Olen, 2022-04-06:
Here is some docs on the plugins:

https://github.com/Olen/solar-monitor/blob/master/PLUGINS.md

excalq, 2022-04-06:
Very nice, I'll practice capturing data, and next time I'm out to the cabin will record what the app sends and receives. It probably doesn't help that the app / equipment are extremely unreliable, needing to be re-paired constantly.

Anto79-ops, 2022-04-06:
@excalq I see some interesting data you picked from the inverter. Interesting.

I wanted to also, to both the connection to the bt1 and bt2 modules. It does improve with Raspberry Pi. So makes you wonder if it's your phone or the DC home app that's causing the connection issues.

I have the current script running on my Raspberry Pi zero 2 w. Things were going well when reading just my rover, it would stay connected to two three weeks at a time, 24/7. Then a contributor to this group added capability for batteries. My RPi was struggling to stay connected for more than an hour or two to both the BT-1 and BT-2. I knew I had a distance issues as there is about 50 feet between them plus through the window and behind my television.

I invested in a long-range usb Bluetooth dongle for my RPi (LM1010-0972), and wow what a difference. Now I can stay connected for at least eight days to both modules without any issues.

So it makes you wonder if it is the app, your phone or the BT modules

excalq, 2022-05-04:
Ok, I was at the cabin over the weekend and got a couple ADB dumps from the official app talking to the Renogy controller over Bluetooth. I'm now looking at them in Wireshark, attempting to make some sense of the capture. (Lots of messages like `Sent Read By Group Type Request, GATT Primary Service Declaration, Handles: 0x002c..0xffff`, which likely map to the Characteristics configs. I'll experiment with these values in a plugin, and see what I come up with.

Regarding connections, since the Pi is inches away from the BT2 module, it seems to get a solid connection. Unsurprisingly, I need to ensure that the Pi and my phone aren't both attempting a connection at the same time. But if I get this working, I'll be happy to use MQTT to store all monitoring data, and never need the app again (which only works if I'm in the battery room anyway.)

Though probably out of scope of this project, I've also purchased a RJ45-to-USB cable from Renogy. None of the Python-based projects to read RS485 have worked though. Solarshed might, but I'm allergic to running a JVM on my already heavily-loaded Pi... But lmk if you have any experience with that.

WWE-Corey, 2022-05-07:
Did you ever get this working? I have a small setup with a BT-1 working fine but built another with the same components as you and get nothing but errors from the BT-2 with a Pi 02W.

```
Debug enabled
DEBUG: Creating new DataLogger
DEBUG: Creating new MQTT-logger
DEBUG: MQTT Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b'sm2'
INFO: Adapter status - Powered: True
INFO: Starting discovery...
DEBUG: MQTT Received CONNACK (0, 0)
DEBUG: Found 6 BLE-devices so far
DEBUG: Found 6 BLE-devices so far
DEBUG: Found 6 BLE-devices so far
DEBUG: Found 6 BLE-devices so far
DEBUG: Found 7 BLE-devices so far
DEBUG: Found 7 BLE-devices so far
DEBUG: Found 7 BLE-devices so far
DEBUG: Found 7 BLE-devices so far
DEBUG: Found 7 BLE-devices so far
INFO: Found 7 BLE-devices
DEBUG: Processing device 60:98:66:f9:34:46 BT-TH-66F93446
INFO: Trying to connect to 60:98:66:f9:34:46...
INFO: Successfully imported SolarLink.
DEBUG: New PowerDevice
DEBUG: New RegulatorDevice
INFO: [regulator] Connecting to 60:98:66:f9:34:46
INFO: [regulator] Connected to BT-TH-66F93446
INFO: [regulator] Resolved services
INFO: [regulator]  Service [f000ffd0-0451-4000-b000-000000000000]
INFO: [regulator]    Characteristic [f000ffd1-0451-4000-b000-000000000000]
INFO: [regulator]  Service [0000fff0-0000-1000-8000-00805f9b34fb]
INFO: [regulator]  - Found dev notify service [0000fff0-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [0000fff1-0000-1000-8000-00805f9b34fb]
INFO: [regulator]  Service [0000ffd0-0000-1000-8000-00805f9b34fb]
INFO: [regulator]  - Found dev write service [0000ffd0-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [0000ffd5-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [0000ffd4-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [0000ffd3-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [0000ffd2-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [0000ffd1-0000-1000-8000-00805f9b34fb]
INFO: [regulator]  Service [0000180a-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [00002a50-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [00002a2a-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [00002a29-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [00002a28-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [00002a27-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [00002a26-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [00002a25-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [00002a24-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [00002a23-0000-1000-8000-00805f9b34fb]
INFO: [regulator] Found dev notify char [0000fff1-0000-1000-8000-00805f9b34fb]
INFO: [regulator] Subscribing to notify char [0000fff1-0000-1000-8000-00805f9b34fb]
INFO: [regulator] Found dev write polling char [0000ffd1-0000-1000-8000-00805f9b34fb]
INFO: [regulator] Found dev write polling char [0000ffd1-0000-1000-8000-00805f9b34fb]
INFO: [regulator] Starting new thread Device-poller-thread regulator
INFO: [regulator] Starting new thread MQTT-poller-thread regulator
DEBUG: [regulator] Looping thread Device-poller-thread regulator
INFO: [regulator] MQTT-poller-thread regulator Waiting for event...
DEBUG: create_poll_request BatteryParamInfo
DEBUG: Processing device 08:6c:32:ba:44:e7 None
DEBUG: create_poll_request BatteryParamInfo => [255, 3, 1, 0, 0, 7, 16, 42]
DEBUG: [regulator] Writing data to 0000ffd1-0000-1000-8000-00805f9b34fb - [255, 3, 1, 0, 0, 7, 16, 42] (ff0301000007102a)
DEBUG: Processing device 00:c3:f4:3e:61:8c None
DEBUG: Processing device 61:75:7f:00:a4:25 Galaxy Watch4 Classic (TBEA)
DEBUG: Processing device 31:d2:a9:8a:ef:f5 None
DEBUG: Processing device 4a:15:e6:4b:fc:c9 None
DEBUG: Processing device 61:0f:bc:6e:66:fe 61-0F-BC-6E-66-FE
INFO: Terminate with Ctrl+C
INFO: [regulator] Notifications enabled for: [0000fff1-0000-1000-8000-00805f9b34fb]
DEBUG: [regulator] Write to characteristic done for: [0000ffd1-0000-1000-8000-00805f9b34fb]
DEBUG: [regulator] Writing data to 0000ffd1-0000-1000-8000-00805f9b34fb - bytearray(b'main recv da ta[20] [') (6d61696e20726563762064612074615b32305d205b)
DEBUG: REG: BatteryParamInfo VAL: b' \x03\xfe'
WARNING: Invalid BS b' \x03\xfe'
WARNING: PollerUpdate - Invalid data: b' \x03\xfe'
DEBUG: [regulator] Writing data to 0000ffd1-0000-1000-8000-00805f9b34fb - bytearray(b'main recv da ta[52] [') (6d61696e20726563762064612074615b35325d205b)
DEBUG: REG: BatteryParamInfo VAL: b'R\xf4\x00\x00\x00\x00\x00\x00\x00\x00\xf9\xfa'
WARNING: Invalid BS (wrong length) b'R\xf4\x00\x00\x00\x00\x00\x00\x00\x00\xf9\xfa'
WARNING: PollerUpdate - Invalid data: b'R\xf4\x00\x00\x00\x00\x00\x00\x00\x00\xf9\xfa'
```

excalq, 2022-05-12:
@WWE-Corey I haven't yet, mostly due to so many other spring projects this property needs, like the irrigation. I'm still in need of getting it working, and I do now have the Wireshark Bluetooth dump, but it's very much a decoding puzzle to solve at this point.

WWE-Corey, 2022-05-13:
FYI I did the irrigation project last year using OpenSprinkler Pi and am pretty happy with it.

Olen, 2023-04-27:
Just some spring cleanup of old issues.

closed 2023-04-27 by Olen

excalq, 2023-04-27:
For anyone reading this in the future, this was unworkable (for me anyway) and the data was too unintelligible. I ended up just adding my own shunt and RS-232 meter (for local power consumption measurement), and a Victron smart shunt for solar-charging/batter monitoring. (It has an onboard MQTT broker). Don't buy Renogy if you value data.