Olen/solar-monitor #15: SmartShunt Support

rhandorf, 2022-05-14:
It seems like this works pretty darn closely with Victron Energy's SmartShunt via Bluetooth. Here's the relevant output from the debug that I have going at the moment.

INFO: Adapter status - Powered: True
INFO: Starting discovery...
INFO: Found 6 BLE-devices
INFO: Trying to connect to [SNIP MAC ADDRESS]...
INFO: Successfully imported RenogyBatt.
INFO: [SmartShunt] Connecting to [SNIP MAC ADDRESS]
INFO: Trying to connect to [SNIP MAC ADDRESS]...
INFO: Successfully imported SolarLink.
INFO: [regulator] Connecting to [SNIP MAC ADDRESS]
INFO: [regulator] Connected to BT-TH-THELASTHALF
INFO: [regulator] Resolved services
INFO: [regulator]  Service [f000ffd0-0451-4000-b000-000000000000]
INFO: [regulator]    Characteristic [f000ffd1-0451-4000-b000-000000000000]
INFO: [regulator]  Service [0000fff0-0000-1000-8000-00805f9b34fb]
INFO: [regulator]  - Found dev notify service [0000fff0-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [0000fff1-0000-1000-8000-00805f9b34fb]
INFO: [regulator]  Service [0000ffd0-0000-1000-8000-00805f9b34fb]
INFO: [regulator]  - Found dev write service [0000ffd0-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [0000ffd5-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [0000ffd4-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [0000ffd3-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [0000ffd2-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [0000ffd1-0000-1000-8000-00805f9b34fb]
INFO: [regulator]  Service [0000180a-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [00002a50-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [00002a2a-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [00002a29-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [00002a28-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [00002a27-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [00002a26-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [00002a25-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [00002a24-0000-1000-8000-00805f9b34fb]
INFO: [regulator]    Characteristic [00002a23-0000-1000-8000-00805f9b34fb]
INFO: [regulator] Found dev notify char [0000fff1-0000-1000-8000-00805f9b34fb]
INFO: [regulator] Subscribing to notify char [0000fff1-0000-1000-8000-00805f9b34fb]
INFO: [regulator] Found dev write polling char [0000ffd1-0000-1000-8000-00805f9b34fb]
INFO: [regulator] Found dev write polling char [0000ffd1-0000-1000-8000-00805f9b34fb]
INFO: [regulator] Starting new thread Device-poller-thread regulator
INFO: [regulator] Starting new thread MQTT-poller-thread regulator
INFO: Terminate with Ctrl+C
INFO: [regulator] MQTT-poller-thread regulator Waiting for event...
INFO: [SNIP MAC ADDRESS] Discovered, alias = SmartShunt HQ2141F332K
INFO: [SmartShunt] Connected to SmartShunt HQ2141F332K
INFO: [regulator] Notifications enabled for: [0000fff1-0000-1000-8000-00805f9b34fb]
INFO: [regulator] Sending new data current: 0.0
INFO: [regulator] Sending new data input_current: 0.0
INFO: [regulator] Sending new data charge_current: 0.0
INFO: [regulator] Sending new data voltage: 0.0
INFO: [regulator] Sending new data input_voltage: 0.0
INFO: [regulator] Sending new data charge_voltage: 13.2
INFO: [regulator] Sending new data power: 0.0
INFO: [regulator] Sending new data input_power: 0.0
INFO: [regulator] Sending new data charge_power: 0.0
INFO: [regulator] Sending new data soc: 98.0
INFO: [regulator] Sending new data capacity: 0.0
INFO: [regulator] Sending new data power_switch: 0
INFO: Creating MQTT-listener solar/regulator/power_switch/set
INFO: [regulator] Sending new data temperature: 22.0
INFO: [regulator] Sending new data battery_temperature: 25.0
INFO: [SNIP MAC ADDRESS] Discovered, alias = SmartShunt HQ2141F332K
INFO: [SmartShunt] Connected to SmartShunt HQ2141F332K
INFO: [SmartShunt] Resolved services
INFO: [SmartShunt]  Service [306b0001-b081-4037-83dc-e59fcc3cdfd0]
INFO: [SmartShunt]    Characteristic [306b0004-b081-4037-83dc-e59fcc3cdfd0]
INFO: [SmartShunt]    Characteristic [306b0003-b081-4037-83dc-e59fcc3cdfd0]
INFO: [SmartShunt]    Characteristic [306b0002-b081-4037-83dc-e59fcc3cdfd0]
INFO: [SmartShunt]  Service [97580001-ddf1-48be-b73e-182664615d8e]
INFO: [SmartShunt]    Characteristic [97580006-ddf1-48be-b73e-182664615d8e]
INFO: [SmartShunt]    Characteristic [97580004-ddf1-48be-b73e-182664615d8e]
INFO: [SmartShunt]    Characteristic [97580003-ddf1-48be-b73e-182664615d8e]
INFO: [SmartShunt]    Characteristic [97580002-ddf1-48be-b73e-182664615d8e]
INFO: [SmartShunt]  Service [68c10001-b17f-4d3a-a290-34ad6499937c]
INFO: [SmartShunt]    Characteristic [68c10003-b17f-4d3a-a290-34ad6499937c]
INFO: [SmartShunt]    Characteristic [68c10002-b17f-4d3a-a290-34ad6499937c]
INFO: [SmartShunt]  Service [00001801-0000-1000-8000-00805f9b34fb]
INFO: [SmartShunt]    Characteristic [00002a05-0000-1000-8000-00805f9b34fb]
INFO: [SmartShunt] Starting new thread Device-poller-thread SmartShunt
Exception in thread Device-poller-thread SmartShunt:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
INFO: [SmartShunt] Starting new thread MQTT-poller-thread SmartShunt
INFO: [SmartShunt] MQTT-poller-thread SmartShunt Waiting for event...
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
INFO: [regulator] Sending new data input_voltage: 13.1
INFO: [regulator] Sending new data input_voltage: 13.0
    self._target(*self._args, **self._kwargs)
  File "/home/rhandorf/solar-monitor/solardevice.py", line 285, in device_poller
WARNING: [regulator] Write to characteristic failed for: [0000ffd1-0000-1000-8000-00805f9b34fb] with error [In Progress]
    self.characteristic_write_value(data, self.device_write_characteristic_polling)
WARNING: [regulator] Write to characteristic failed for: [0000ffd1-0000-1000-8000-00805f9b34fb] with error [In Progress]
  File "/home/rhandorf/solar-monitor/solardevice.py", line 255, in characteristic_write_value
WARNING: [regulator] Write to characteristic failed for: [0000ffd1-0000-1000-8000-00805f9b34fb] with error [In Progress]
    logging.debug("[{}] Writing data to {} - {} ({})".format(self.logger_name, write_characteristic.uuid, value, bytearray(value).hex()))
AttributeError: 'NoneType' object has no attribute 'uuid'
WARNING: [regulator] Write to characteristic failed for: [0000ffd1-0000-1000-8000-00805f9b34fb] with error [In Progress]
WARNING: [regulator] Write to characteristic failed for: [0000ffd1-0000-1000-8000-00805f9b34fb] with error [In Progress]
WARNING: [regulator] Write to characteristic failed for: [0000ffd1-0000-1000-8000-00805f9b34fb] with error [In Progress]
INFO: [regulator] Sending new data soc: 97.0
INFO: [SNIP MAC ADDRESS] Discovered, alias = SmartShunt HQ2141F332K
INFO: [SmartShunt] Disconnected
INFO: [SmartShunt] Stopping poller-thread
INFO: [SmartShunt] Stopping command-thread
INFO: [SmartShunt] MQTT-poller-thread SmartShunt Event happened...
ERROR: [SmartShunt] MQTT-poller-thread SmartShunt Something bad happened: 'SmartShunt'
INFO: [SmartShunt] Ending thread MQTT-poller-thread SmartShunt
INFO: [regulator] Sending new data soc: 98.0

Any suggestions as to how I can try and help out to get it added?

Thanks!

rhandorf, 2022-05-19:
Here's some additional resourcing about the endpoints: https://community.victronenergy.com/questions/93919/victron-bluetooth-ble-protocol-publication.html

and I found these two projects as well: 
https://github.com/birdie1/victron
https://github.com/vvvrrooomm/victron/

Olen, 2023-04-27:
Just some spring cleaning of old issues

closed 2023-04-27 by Olen