Olen/solar-monitor #17: Renogy Rover History, generation and min max support

Anto79-ops, 2022-05-16:
Hi.

The for the Renogy Rover CC, it would be great if the script can also report history, daily generation (kWh), and min and max values

thanks

Anto79-ops, 2022-05-26:
@Olen you raise some good points. However, good news is that the Rover has some sort of data saving chip and sends this values out like a regular number.

Even better, a fellow here, @oedo808 has already done the leg work with a fork of your script and sniffed out the values reponsible for this, here:

https://github.com/oedo808/solar-monitor

At the moment, the values are being input into the debug files. so the datalogger/mqtt parts are not set up. This requires some more work, but its almost there if anyone has the skills and time to do the latter parts.

oedo808, 2022-05-26:
I'll try to get this knocked out.  I started splitting a jumbo branch up into individual feature branches to make it easier to roll back but I'm awful with rewriting commits.  I'll try to just finish the datalogger work in the jumbo branch and submit a PR rather than trying to organize the git mess I made.

oedo808, 2022-05-26:
Also, I've been running with it grabbing historical data in debug mode regularly for a while.  It doesn't seem to impact much, it's just an extra set of data points, maybe double the size of the normal realtime data.  

I've had some issues since, but I think they're unrelated to historical data in general and more towards filling up my tiny 8GB SD card with all of the journalctl debug logging from the solarmonitor service.

Ideally the historical data should probably be polled daily or at least 8 hours, but I think it's reasonably stable polling regularly.

Olen, 2022-05-26:
Just out of curiosity - do the historical data provide you with any info that you can not get if you log data to a database?
If I already get all the live updates from the devices regularily, why would I also want the historical data?

Anto79-ops, 2022-06-23:
probably you can get the same if you had your own script or graph program to extract that from the data provided, but the Rover does do that for you, charge status would also be good to know (i.e. boost, mptt, float, etc).

closed 2024-07-23 by Anto79-ops