Olen/solar-monitor #18: Renogy Rover 20A MPPT issues

Byter09, 2022-05-16:
Hello.

Because I don't like running software bare-metal on my server, I decided to write a docker image.

```Dockerfile
FROM archlinux

RUN pacman -Syyuu --noconfirm
RUN pacman -S git python-pip python-setuptools python-dbus python-gobject nano --noconfirm

RUN mkdir /solar-monitor
WORKDIR /solar-monitor
RUN git clone https://github.com/Olen/solar-monitor.git .
RUN pip install -r requirements.txt
```

So far so good, that builds perfectly.

I then launch this using docker compose:
```yml
services:
  solar-monitor:
    image: solar-monitor:latest
    container_name: solar-monitor
    environment:
      PUID: 1000
      PGID: 1001
      TZ: Europe/Berlin
    volumes:
      - /local/path/to/solar-monitor.ini:/solar-monitor/solar-monitor.ini
      - /var/run/dbus:/var/run/dbus
    #restart: unless-stopped # once I get this working, I'll turn this on
    privileged: true
    #tty: true # debug purposes
    command: bash -c "python solar-monitor.py"

networks:
  default:
    external:
      name: web

```

This, as expected, also works. So far I'm fairly certain it is _not_ my set up that is the problem and rather a software bug.

Here's the configuration file:
```ini
[monitor]
adapter = hci0
debug = True
temperature = C
reconnect = False
# C = Celsius
# K = Kelvin
# F = Farenheit

[renogy_rover_20A]
type = RenogyBatt
mac = E0:7D:EA:76:98:07
reconnect = True

[datalogger]
url = https://my.domain/webhook-test/solar
token = changeme

```

I was unsure if the Renogy Rover 20A is going to be supported, but it seems to communicate fine, till it runs into a problem that I sadly have no clue about what could be wrong.
```
DEBUG: Creating new DataLogger
INFO: Adapter status - Powered: True
INFO: Starting discovery...
DEBUG: Found 1 BLE-devices so far
DEBUG: Found 1 BLE-devices so far
DEBUG: Found 1 BLE-devices so far
DEBUG: Found 1 BLE-devices so far
DEBUG: Found 1 BLE-devices so far
DEBUG: Found 1 BLE-devices so far
INFO: Found 1 BLE-devices
DEBUG: Processing device e0:7d:ea:76:98:07 BT-TH-EA769807
INFO: Trying to connect to e0:7d:ea:76:98:07...
INFO: Successfully imported RenogyBatt.
DEBUG: New PowerDevice
INFO: [renogy_rover_20A] Connecting to e0:7d:ea:76:98:07
INFO: [renogy_rover_20A] Connected to BT-TH-EA769807
INFO: [renogy_rover_20A] Resolved services
INFO: [renogy_rover_20A]  Service [f000ffd0-0451-4000-b000-000000000000]
INFO: [renogy_rover_20A]    Characteristic [f000ffd1-0451-4000-b000-000000000000]
INFO: [renogy_rover_20A]  Service [0000fff0-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]  - Found dev notify service [0000fff0-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [0000fff1-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]  Service [0000ffd0-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]  - Found dev write service [0000ffd0-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [0000ffd5-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [0000ffd4-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [0000ffd3-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [0000ffd2-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [0000ffd1-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]  Service [0000180a-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [00002a50-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [00002a2a-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [00002a29-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [00002a28-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [00002a27-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [00002a26-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [00002a25-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [00002a24-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [00002a23-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A] Found dev notify char [0000fff1-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A] Subscribing to notify char [0000fff1-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A] Found dev write polling char [0000ffd1-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A] Found dev write polling char [0000ffd1-0000-1000-8000-00805f9b34fb]
Debug enabled
Traceback (most recent call last):
  File "/solar-monitor/solar-monitor.py", line 100, in <module>
INFO: [renogy_rover_20A] Starting new thread Device-poller-thread renogy_rover_20A
DEBUG: [renogy_rover_20A] Looping thread Device-poller-thread renogy_rover_20A
    device.connect()
DEBUG: create_poll_request TotalCapacity
  File "/solar-monitor/solardevice.py", line 105, in connect
DEBUG: create_poll_request TotalCapacity => [48, 3, 20, 2, 0, 8, 228, 29]
    super().connect()
DEBUG: [renogy_rover_20A] Writing data to 0000ffd1-0000-1000-8000-00805f9b34fb - [48, 3, 20, 2, 0, 8, 228, 29] (300314020008e41d)
  File "/usr/lib/python3.10/site-packages/gatt/gatt_linux.py", line 288, in connect
    self._connect()
  File "/usr/lib/python3.10/site-packages/gatt/gatt_linux.py", line 295, in _connect
    self.services_resolved()
  File "/solar-monitor/solardevice.py", line 187, in services_resolved
    self.datalogger.mqtt.trigger[self.logger_name] = self.command_trigger
AttributeError: 'NoneType' object has no attribute 'trigger'
```

I don't quite understand how I can get an error related to MQTT if I don't even want to use it. All I want is for `solar-monitor` to send the data to a webhook.

If I missed some information that you need to better identify what I'm doing wrong, let me know! I'll play around with all sorts of settings if need be.

Olen, 2022-05-23:
This should be fixed now.  Please pull the latest changes and try again.

Byter09, 2022-05-23:
Tried it. Same config as before. Debug mode shows me getting temperature, current, etc etc, but I never received data on the webhook. At least its not crashing now :D

Between creating this issue and now I used https://github.com/cyrils/renogy-bt1 and wrote my own code to push it to a webhook. I get data roughly every 5 seconds with this one.

So unsure if I'm doing something wrong with solar-monitor :/

Olen, 2022-05-24:
How does your .ini-file look (especially the `[datalogger]` part?
What does the log say when a value is updated?
It should say saomething like
```
Sending new data: xxxx: yyyy
```

I also added some extra logging til the webhook-part, so it should now (with the latest commit) also log something like
```
Sending data to <your_url>
```

Byter09, 2022-05-25:
I posted the configuration file in the issue, below the docker compose.

And it never actually sends to my webhook. All I get is this:

```
[root@9389b8f879c7 solar-monitor]# git pull
Already up to date.
[root@9389b8f879c7 solar-monitor]# python solar-monitor.py
Debug enabled
DEBUG: Creating new DataLogger
INFO: Adapter status - Powered: True
INFO: Starting discovery...
DEBUG: Found 1 BLE-devices so far
DEBUG: Found 1 BLE-devices so far
DEBUG: Found 1 BLE-devices so far
DEBUG: Found 1 BLE-devices so far
DEBUG: Found 1 BLE-devices so far
DEBUG: Found 1 BLE-devices so far
INFO: Found 1 BLE-devices
DEBUG: Processing device e0:7d:ea:76:98:07 BT-TH-EA769807
INFO: Trying to connect to e0:7d:ea:76:98:07...
INFO: Successfully imported RenogyBatt.
DEBUG: New PowerDevice
INFO: [renogy_rover_20A] Connecting to e0:7d:ea:76:98:07
INFO: [renogy_rover_20A] Connected to BT-TH-EA769807
INFO: [renogy_rover_20A] Resolved services
INFO: [renogy_rover_20A]  Service [f000ffd0-0451-4000-b000-000000000000]
INFO: [renogy_rover_20A]    Characteristic [f000ffd1-0451-4000-b000-000000000000]
INFO: [renogy_rover_20A]  Service [0000fff0-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]  - Found dev notify service [0000fff0-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [0000fff1-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]  Service [0000ffd0-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]  - Found dev write service [0000ffd0-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [0000ffd5-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [0000ffd4-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [0000ffd3-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [0000ffd2-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [0000ffd1-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]  Service [0000180a-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [00002a50-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [00002a2a-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [00002a29-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [00002a28-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [00002a27-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [00002a26-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [00002a25-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [00002a24-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A]    Characteristic [00002a23-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A] Found dev notify char [0000fff1-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A] Subscribing to notify char [0000fff1-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A] Found dev write polling char [0000ffd1-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A] Found dev write polling char [0000ffd1-0000-1000-8000-00805f9b34fb]
INFO: [renogy_rover_20A] Starting new thread Device-poller-thread renogy_rover_20A
INFO: Terminate with Ctrl+C
DEBUG: [renogy_rover_20A] Looping thread Device-poller-thread renogy_rover_20A
DEBUG: create_poll_request TotalCapacity
DEBUG: create_poll_request TotalCapacity => [48, 3, 20, 2, 0, 8, 228, 29]
DEBUG: [renogy_rover_20A] Writing data to 0000ffd1-0000-1000-8000-00805f9b34fb - [48, 3, 20, 2, 0, 8, 228, 29] (300314020008e41d)
INFO: [renogy_rover_20A] Notifications enabled for: [0000fff1-0000-1000-8000-00805f9b34fb]
DEBUG: [renogy_rover_20A] Write to characteristic done for: [0000ffd1-0000-1000-8000-00805f9b34fb]
DEBUG: [renogy_rover_20A] Looping thread Device-poller-thread renogy_rover_20A
DEBUG: [renogy_rover_20A] Looping thread Device-poller-thread renogy_rover_20A
DEBUG: create_poll_request VoltageCurrentSOC
DEBUG: create_poll_request VoltageCurrentSOC => [48, 3, 19, 178, 0, 6, 101, 74]
DEBUG: [renogy_rover_20A] Writing data to 0000ffd1-0000-1000-8000-00805f9b34fb - [48, 3, 19, 178, 0, 6, 101, 74] (300313b20006654a)
DEBUG: [renogy_rover_20A] Write to characteristic done for: [0000ffd1-0000-1000-8000-00805f9b34fb]
DEBUG: [renogy_rover_20A] Looping thread Device-poller-thread renogy_rover_20A
DEBUG: [renogy_rover_20A] Looping thread Device-poller-thread renogy_rover_20A
DEBUG: create_poll_request CellVoltage
DEBUG: create_poll_request CellVoltage => [48, 3, 19, 136, 0, 17, 5, 73]
[... 47 lines cut ...]
DEBUG: [renogy_rover_20A] Write to characteristic done for: [0000ffd1-0000-1000-8000-00805f9b34fb]
DEBUG: [renogy_rover_20A] Looping thread Device-poller-thread renogy_rover_20A
DEBUG: create_poll_request VoltageCurrentSOC
DEBUG: create_poll_request VoltageCurrentSOC => [48, 3, 19, 178, 0, 6, 101, 74]
DEBUG: [renogy_rover_20A] Writing data to 0000ffd1-0000-1000-8000-00805f9b34fb - [48, 3, 19, 178, 0, 6, 101, 74] (300313b20006654a)
DEBUG: [renogy_rover_20A] Write to characteristic done for: [0000ffd1-0000-1000-8000-00805f9b34fb]
DEBUG: [renogy_rover_20A] Looping thread Device-poller-thread renogy_rover_20A
DEBUG: create_poll_request Temperature
DEBUG: create_poll_request Temperature => [48, 3, 19, 153, 0, 7, 212, 130]
DEBUG: [renogy_rover_20A] Writing data to 0000ffd1-0000-1000-8000-00805f9b34fb - [48, 3, 19, 153, 0, 7, 212, 130] (300313990007d482)
DEBUG: [renogy_rover_20A] Write to characteristic done for: [0000ffd1-0000-1000-8000-00805f9b34fb]
[... cut ...]
DEBUG: [renogy_rover_20A] Looping thread Device-poller-thread renogy_rover_20A
^CDEBUG: [renogy_rover_20A] Looping thread Device-poller-thread renogy_rover_20A
DEBUG: create_poll_request VoltageCurrentSOC
DEBUG: create_poll_request VoltageCurrentSOC => [48, 3, 19, 178, 0, 6, 101, 74]
DEBUG: [renogy_rover_20A] Writing data to 0000ffd1-0000-1000-8000-00805f9b34fb - [48, 3, 19, 178, 0, 6, 101, 74] (300313b20006654a)
DEBUG: [renogy_rover_20A] Looping thread Device-poller-thread renogy_rover_20A
```
... and so on, and so on. It simply loops forever like this.

Anto79-ops, 2022-05-25:
i could be mistaken...but why is it importing "INFO: Successfully imported RenogyBatt" when you are using the Rover? The Rover should be solarlink/regulator.

you should be putiing your mac address here for your Rover20

[regulator]
type = SolarLink
mac = 11:11:11:11:11:11
reconnect = False

Byter09, 2022-05-25:
> The Rover should be solarlink/regulator.

See I didn't know that. Did I miss the documentation on that somewhere? Gonna give it a shot.

Anto79-ops, 2022-05-25:
No problem, this is how we learn. This should solve your problem.

RenogyBatt should be used for Renogy Batteries
Solarlink/regulator should be use for Renogy Rovers

Byter09, 2022-05-25:
Alrighty, next problem: "state" is not being transmitted. The log states "Could not find state", where as https://github.com/cyrils/renogy-bt1 successfully reports the state like "mppt", "off" or "boost".

Could it be that using `SolarLink` is not _quite_ right here?

Byter09, 2022-05-25:
And one more. The current is not being transmitted correctly.
Data says 1.39A, but is sending 1.4.
```
DEBUG: mLoadElectricity 0 139 => 1.3900000000000001 A
DEBUG: [renogy_rover_20A] Value of _mcurrent changed from 0 to 1390.0000000000002
INFO: [renogy_rover_20A] Sending new data current: 1.4
```

Olen, 2022-05-26:
> Could it be that using SolarLink is not quite right here?

Could be. I don't have the Rover, so I don't know much about it.  Please look at e.g. #7 on how to help getting the correct values from devices. 

> Data says 1.39A, but is sending 1.4.

This is called "rounding".  
To balance how much data is sent, updates with just tiny variations are not published, as this would flood the receiver with useless data.

Byter09, 2022-05-26:
I'm aware what rounding is and I understand why this would help in reduce data being sent, but assuming the value is 1.38, and you're sending 1.4, that means over time you're constantly missing those 0.02 A. That may not sound like much, but if you want to do accurate calculations like I do, this is incredibly important information and should not be rounded away. Worst case, a settings options to disable rounding would be appreciated.

And I'll check out #7. All I was saying is that there is a script on GitHub that I linked, that already gets this data correctly, which is likely something one can simply copy 🤷‍♂️

Olen, 2022-05-26:
I am happy to accept a PR both for a config-switch regarding rounding and for the support of more devices.
At least my regulator is sending data many times a second with every minor change in all the values, so not doing any rounding means way to much traffic to the receiver (this is running over 4G) and will not really gain any value, as it will just as often be 1.42A as 1.38A, and these fluctations will even out over time.

But again.  Please submit a PR, and I'll have a look at it.

Byter09, 2022-05-26:
I don't really dabble in python but I'm on vacation so I might dive in haha.

Olen, 2022-05-26:
Btw. it seems like the script you linked to is heavily based on this code, so you could try to convice @cyrils to help you or submit his fixes back here.

Olen, 2023-04-27:
Just some spring cleaning of old issues

closed 2023-04-27 by Olen