Olen/solar-monitor #21: How to install on RPi4 running HA OS?

philjmaier, 2022-08-18:
@Olen , is it possible to install this on a Home Assistant OS installation?

Olen, 2022-08-19:
It should be possible.  Maybe not on HA OS as I don't know how much control you have over third party applications there, but at least with HA docker.  You could get an issue with HA and this fighting over control of the bluetooth device, so I can't make any promises.

In a world of unlimited time, I would sit down and create this as a HA integration, using the new bluetooth module in HA, but unfortunately I don't think that is goingto be a priority.  
I run this on a standalone RPI and send the data with MQTT to HA.

Olen, 2023-01-05:
I just created a very simple Dockerfile.
It seems to work, but you need to mount your solar-monitor.ini inside the container and also mount /var/run/dbus

I'll improve it and document the usage.

closed 2023-04-27 by Olen