Olen/solar-monitor #25: MQTT retain - entities in HA

Anto79-ops, 2023-01-03:
Hey, not sure if this a code related or HASS discovery related, but every time I have to restart my RPi for updates and resume the script, the entities get all changed up in HASS, for example it may add or remote a  "_2" at the end of the entity, and it kind of messes up my automations.

How can the names be retained on restart? thanks

Olen, 2023-01-05:
Not sure why this happens to you.  I have restarted lots of times, and just very rarely I have gotten a "_2" entity or two, but more than 99% of the times, all the entities are retained.

It could be somwthing with your MQTT settings maybe?

Anto79-ops, 2023-02-17:
Hi, I just wanted to follow up on this, is the code using [asyncio-mqtt](https://sbtinstruments.github.io/asyncio-mqtt/publishing-and-subscribing.html)?

if so, then retaining the mqtt entities after restart can be achieved by simply placing` retain=True` to any `client.publish` calls.

Olen, 2023-02-17:
It is using paho
https://github.com/Olen/solar-monitor/blob/c263d350b23f44f2278b1cd90ccb5a07050e95af/datalogger.py#L6

And retain is set
https://github.com/Olen/solar-monitor/blob/c263d350b23f44f2278b1cd90ccb5a07050e95af/datalogger.py#L64

https://github.com/Olen/solar-monitor/blob/c263d350b23f44f2278b1cd90ccb5a07050e95af/datalogger.py#L78

https://github.com/Olen/solar-monitor/blob/c263d350b23f44f2278b1cd90ccb5a07050e95af/datalogger.py#L119

You can also enable debug-logging to see what it receives from the MQTT-server
https://github.com/Olen/solar-monitor/blob/c263d350b23f44f2278b1cd90ccb5a07050e95af/datalogger.py#L154

Anto79-ops, 2023-02-17:
its interesting that the retain is set to true already, now I'm wondering if this has anything to do with HASS discovery.

I'll have to confirm that with the debug-logging

Anto79-ops, 2023-04-13:
hey I'm wondering if this PR will solve this issue here:

https://github.com/home-assistant/core/pull/90801

according to the devs comment:

"...This will prevent an extra async call if for each discovery update that has not changed."

This should be in 2023.5

Anto79-ops, 2023-06-24:
This is not an issue anymore since the update above.

closed 2023-06-24 by Anto79-ops