Olen/solar-monitor #26: system update broke my script

Anto79-ops, 2023-01-26:
Hi all,

I think the last update broke something on my system. When I try to run the solar-monitor, I get this error message:

```
pi@raspberrypi:~/solar-monitor $ python3 solar-monitor.py
INFO: Adapter status - Powered: True
INFO: Starting discovery...
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/gatt/gatt_linux.py", line 156, in stop_discovery
    self._adapter.StopDiscovery()
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 141, in __call__
    return self._connection.call_blocking(self._named_service,
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownObject: Method "StopDiscovery" with signature "" on interface "org.bluez.Adapter1" doesn't exist

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/solar-monitor/solar-monitor.py", line 84, in <module>
    device_manager.stop_discovery()
  File "/home/pi/.local/lib/python3.9/site-packages/gatt/gatt_linux.py", line 161, in stop_discovery
    raise _error_from_dbus_error(e)
gatt.errors.Failed: Method "StopDiscovery" with signature "" on interface "org.bluez.Adapter1" doesn't exist

```

Things have been running fine for almost 2 years now on my RPi Zero 2 W on Bullseye with Long Range USB adapter and ethernet adapter (internal RPi Wifi and BT have been disabled).

not sure where to begin with this one... any insight would be appreciated.

Olen, 2023-01-26:
I guess there is lots of old and fragile code here that might not play well with the latest python versions.

I have long planned making a simple docker container which should at least mitigate the problem temporarily.

It could also be a simple fix, maybe an updated version of some dependency.

Anto79-ops, 2023-01-27:
Yeah, who knows. I tried all day yesterday to get back up and running, but I'm still down.

I completely wiped my RPi and started from scratch, and installed all the libraries again in the requirements file (FYI, about 3 of the libraries kept failing on build, but then after some research I found they were dependent on other libraries to be installed, and then eventually everything went smoothly and all the required libs were installed).

Unfortunately, the issue is still the same. If keeps trying to start the script, eventually the error message turns into this:

```
pi@raspberrypi:~/solar-monitor $ python3 solar-monitor.py
Traceback (most recent call last):
  File "/home/pi/solar-monitor/solar-monitor.py", line 53, in <module>
    [logging.info](http://logging.info/)("Adapter status - Powered: {}".format(device_manager.is_adapter_powered))
  File "/home/pi/.local/lib/python3.9/site-packages/gatt/gatt_linux.py", line 53, in is_adapter_powered
    return self._adapter_properties.Get('org.bluez.Adapter1', 'Powered') == 1
  File "/home/pi/.local/lib/python3.9/site-packages/dbus/proxies.py", line 72, in __call__
    return self._proxy_method(*args, **keywords)
  File "/home/pi/.local/lib/python3.9/site-packages/dbus/proxies.py", line 141, in __call__
    return self._connection.call_blocking(self._named_service,
  File "/home/pi/.local/lib/python3.9/site-packages/dbus/connection.py", line 634, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownObject: Method "Get" with signature "ss" on interface "org.freedesktop.DBus.Properties" doesn't exist
```

So, if anyone comes across a solution....I'm all ears! Sorry, I don't have the skills to submit a PR.

If only I can downgrade the library that caused this, that would be a solution, too!

It really sucks to be down...now I have to rely on the manufactures app, which is not that great in terms of convenience.

Olen, 2023-01-30:
I would try to run it as a docker container instead.  The Dockerfile and docker-compose.yaml provided s working fine here on a rpi-zero.  So you should not have any library/dependency issues with that.

Anto79-ops, 2023-01-31:
all of a sudden, it just starts working. My setup has always been clunky to start, but once it starts, its pretty stable (i.e. can stay connected for weeks to months).

I'm on a raspberry pi forum, asking for some more insight to this. Might be a "timing issue", but I was told to try 'dmesg' when it happens again. I can report back if the findings lead to this specific code, but will close this now.

ITs good to know if al else fails, I could try the dockerfile method. Thanks for sharing that!

closed 2023-01-31 by Anto79-ops