Olen/solar-monitor #32: using solar monitor with Renogy ONE Core

Anto79-ops, 2023-05-26:
hey!

Wanted to pink you @acmagn. I was given a Renogy ONE Core to review. Its Renogy's new hub and monitoring center. One great thing about this, is that it has an ethernet port for RS-485 communication. I thought that I can use both  Renogy ONE Core and Solar-monitor at the same time, so I still have my data/automation in Home Assistant.

To my surprise, it works! I can use both wired (for DC Home) and BT connection (for HA) at the same time.

Something strange thought ha happened to my cell 1 to 3 voltages, its seems something crossing with temperature(?) if that makes sense. For example, cell voltage should be 3.6 V it does report 3.6 V but it also flips to 22 V which is suspiciously the same as the temperature, for example:

[image]

Now, I do have another Renogy product on the hub and seems to cause quite a bit extra traffic (lots of warning lines on the CLI as the script says something is not supported, its their Renogy Pure Sine Wave inverter with communication via RS-485).

anyways, here's some logs that I just quickly pasted:

```

NG: [renogy_battery_1] Write to characteristic failed for: [0000ffd1-0000-1000-8000-00805f9b34fb] with error [In Progress]
WARNING: [renogy_battery_1] Write to characteristic failed for: [0000ffd1-0000-1000-8000-00805f9b34fb] with error [In Progress]
WARNING: Invalid BS (wrong length) b'0\x03\x13\xb2\x00\x01$\x88'
WARNING: PollerUpdate - Invalid data: b'0\x03\x13\xb2\x00\x01$\x88'
WARNING: CRC Failed: 16241 - Check: 63488
WARNING: PollerUpdate - Invalid data: b'0\x03\x00\x00\xf8'
WARNING: [renogy_battery_1] Write to characteristic failed for: [0000ffd1-0000-1000-8000-00805f9b34fb] with error [In Progress]
WARNING: Invalid BS (wrong length) b'0\x03\x13\xb3\x00\x01uH'
WARNING: PollerUpdate - Invalid data: b'0\x03\x13\xb3\x00\x01uH'
WARNING: [renogy_battery_1] Write to characteristic failed for: [0000ffd1-0000-1000-8000-00805f9b34fb] with error [In Progress]
WARNING: Invalid BS (wrong length) b'0\x03\x13\x88\x00\x11\x05I'
WARNING: PollerUpdate - Invalid data: b'0\x03\x13\x88\x00\x11\x05I'
WARNING: [renogy_battery_1] Write to characteristic failed for: [0000ffd1-0000-1000-8000-00805f9b34fb] with error [In Progress]
WARNING: Invalid BS (wrong length) b'0\x03"\x00\x04\x00$\x00$\x00$\x00$\x00\x00'
WARNING: PollerUpdate - Invalid data: b'0\x03"\x00\x04\x00$\x00$\x00$\x00$\x00\x00'
WARNING: Invalid BS (wrong length) b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc1\xe4'
WARNING: PollerUpdate - Invalid data: b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc1\xe4'
WARNING: [renogy_battery_1] Write to characteristic failed for: [0000ffd1-0000-1000-8000-00805f9b34fb] with error [In Progress]
WARNING: Invalid BS (wrong length) b'0\x03\x13\x99\x00\x11UL'
WARNING: PollerUpdate - Invalid data: b'0\x03\x13\x99\x00\x11UL'
WARNING: [renogy_battery_1] Write to characteristic failed for: [0000ffd1-0000-1000-8000-00805f9b34fb] with error [In Progress]
INFO: [renogy_battery_1] Sending new data cell_1: 22000
INFO: [renogy_battery_1] Sending new data cell_2: 21000
INFO: [renogy_battery_1] Sending new data cell_3: 21000
INFO: [renogy_battery_1] Sending new data cell_1_voltage: 22.0
INFO: [renogy_battery_1] Sending new data cell_2_voltage: 21.0
INFO: [renogy_battery_1] Sending new data cell_3_voltage: 21.0
WARNING: Invalid BS (wrong length) b'0\x03\x13\xec\x00\x06\x04\x18'
WARNING: PollerUpdate - Invalid data: b'0\x03\x13\xec\x00\x06\x04\x18'
WARNING: [renogy_battery_1] Write to characteristic failed for: [0000ffd1-0000-1000-8000-00805f9b34fb] with error [In Progress]
WARNING: Invalid BS (wrong length) b'0\x03\x13\xf2\x00\x01%\\'
WARNING: PollerUpdate - Invalid data: b'0\x03\x13\xf2\x00\x01%\\'
```

it seems to be only Cell voltage 1 to 3. Any idea why these cell voltage are kinda of flipping with this new setup?

The inverter definitely could be interfering, but not sure why it is now an issue. I guess the BT2 module now just spit everything out to the ONE Core so the script is picking it up.

I wonder if I can catch logs, maybe the Inverter can be added the script as another device to add at some point, but why its contaminating with the battery cell voltages is my main question. Any idea?

acmagn, 2023-07-16:
Hey @Anto79-ops! I've been keeping an eye on that Renogy ONE Core, does it provide historical data?

I had to think on this one a bit, but I have a guess to what is happening:
There are ways to write data to the Renogy modules to change what they output. In testing, I was able to change what the battery returned. However at the time I didn't look too deeply into what these other values were.
I think the Renogy ONE core is requesting reports on the cell temperatures, flipping an internal switch in the batteries, and naturally the solar monitor is also receiving that new data.

acmagn, 2023-07-16:
It could be possible to add the inverter at some time, hard to estimate the work involved without me owning one. It would probably be a good amount of a back-and-forth. If I'm feeling up for it at some point I could potentially buy one, get it working with the monitor, and then return it haha.

Anto79-ops, 2023-07-20:
thanks @acmagn althought this works I decided not to keep the ONE Core running because it dumps so much extra information on the bus that the script fails after about a 1 day, so now I just the Core as a backup.

By the way, someone posted that they were able to figure out the api for these new Core devices....but the comment has since been deleted so I cannot share it here 👎

But, I guess its possible to reverse engineer those IOT devices

closed 2023-07-20 by Anto79-ops