Olen/solar-monitor #35: ERROR: Unable to set up datalogger ERROR: No option 'hostname' in section: 'mqtt'

sclawer, 2024-06-04:
Hi there
Can you please provide some greater detail as to how to setup MQTT 

I have the following within my solar-monitor.ini file

`
#[datalogger]
#url = http://server/solar/api/
#token = 39129e20be0503937cb72a5f719337cc

[mqtt]
# Address to the mqtt broker
broker = 192.168.1.26
# All topics are prefixed by this prefix
# E.g. "solar/battery_1/voltage"
prefix = solar
username = username
password = password`

I have glanced the code an guess I have the credentials sorted.. but the above error is presenting

sclawer, 2024-06-04:
FYI 
Tried broker = IP:8123
tried hostname = IP -> different error

Using RasPi 4B - running solar-monitor in own docker container, with mosquitto in another container - can send and receive messages from HAAS and my machine

Olen, 2024-06-05:
Sorry, I'll update the docs.

`broker` is the MQTT-broker
`hostname` is used in the mqtt-connection and should match the local hostname

Both `prefix` and `hostname` should be optional, but I see there is a problem in the datalogger, where there is no fallback configured if it is not set. 

I'll create a patch for this

sclawer, 2024-06-05:
Thank you, this issue has been resolved

closed 2024-06-05 by sclawer